FreeOpcUa/async-opcua #4: Rename `browser/implementation.rs` and `session/implementation.rs`?

svanharmelen, 2024-12-19:
Of course this is just nitpicking and has no actual value, but I was surprised to see `session/session.rs` being renamed to `session/implementation.rs` and noticed the same pattern in the `browse` module.

It's not a layout I'm used to and/or see much in the Rust community so wondered if you are open to renaming them (back) to `browser/browser.rs` and `session/session.rs`?

While I understand it's kind of a stutter to have these repeated names, but personally I find it easier and more intuitive to navigate code when the filenames describe the main thing you can find in that file 🤷🏻‍♂️

einarmo, 2024-12-19:
I believe there's a default clippy lint against `module.rs` inside `module`, which is why I avoid them... I can agree there's a better way to name these though.

einarmo, 2024-12-19:
Indeed there is: https://rust-lang.github.io/rust-clippy/master/index.html#module_inception.

That said, we can clean it up a different way. For `session` we could move that into `mod.rs`. For `browser`, `implementation.rs` actually _does_ contain the implementation of the browser. Other possible names could be `run.rs` or `execute.rs`, or something along those lines.

einarmo, 2024-12-22:
Done in #7

closed 2024-12-22 by einarmo