FreeOpcUa/async-opcua #27: Give the library a name

einarmo, 2025-01-10:
Now that we've had some time to iron out any obvious issues, and I feel at least some people have tried the library, I think it's about time we create a release process and release to crates.io.

This means renaming the crates, so we need to find one, then make the effort of changing the code to respect the new crate names.

My only real requirement is that I want the crate name to contain `opcua` still, since I think that's good for discoverability. My initial thoughts are something like `opcua-oxide` or `opcua-rs`, but I'm open for suggestions. Do remember that this affects all our crates, so we end up with `opcua-oxide-types` and `opcua-oxide-base-namespace` and so on.

oroulet, 2025-01-10:
async-opcua was proposed there: https://github.com/FreeOpcUa/rust-opcua/discussions/19 since async might be the biggest change

einarmo, 2025-01-11:
Did not see that discussion, it is probably a more natural place

closed 2025-01-11 by einarmo