FreeOpcUa/async-opcua #29: Cannot generate EUROMAP NodeSet

evanjs, 2025-01-21:
## Cannot generate EUROMAP NodeSets

✅ generate types for EUROMAP 79, 82.2, 83
🚫 generate NodeSets for EUROMAP 83
___
## Foreward
Please pardon the wall of text, I hope there is at least some helpful information in here

I wanted to persist my thought on using this library before I am fully occupied fixing the OPCUA application that currently utilizes https://github.com/locka99/opcua

There's also a non-zero chance that I am simply not configuring code generation correctly
Knowing how to resolve that would be fine, of course.
___
# Issue Details

Hey guys, thanks for spinning up this fork, I've been meaning to try out the changes for async restructuring, but haven't had a chance.

Currently I am working on a project that is already being deployed but uses the original repository.
Quite a few things needed to be worked around, but I'm excited to see how this library could make things easier for me.

When I was trying the repository out the other day, I noticed that code generation seems to have been reworked and now does a lot more than it did previously.

In my case, I was able to generate types for the EUROMAP specs I have been using.
If I understand correctly, this is a new feature (BSD support) compared to the original repository

However, when attempting to generate nodesets using said types (using documents from the same specifications), the process failed with an XmlError (MissingContent).

As noted in the different design philosophy section of the documentation (under [fork/Different Design Philosophy](https://github.com/FreeOpcUa/rust-opcua/blob/master/docs/fork.md#different-design-philosophy)), it seems that this repository seems to be a bit more focused in a way that the original was not.

My interpretation of this is that we probably don't want to add, say, generated EUROMAP types, etc. directly to this repository in the form of generated code.

In spite of that, I am hoping we can ensure that this library can still be used to generate all (or most, I have no idea what's feasible) OPCUA standards

Or even create a repository for the generated EUROMAP types/nodesets, as suggested in the aforementioned fork document:
> We will not include support for any companion standards, but we are open for changes that make it easier to write extension libraries defining companion standards.
> We may even be open for publishing those libraries from here, just not as part of the main library.

___

## Configuration Details

<summary><h1>Codegen Configuration</h1></summary>
<details>

```yml
extra_header: |
  // OPCUA for Rust
  // SPDX-License-Identifier: MPL-2.0
  // Copyright (C) 2017-2024 Einar Omang
targets:
  - type: types
    file_path: resources/OPCUA/EUROMAP/BSD/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.bsd
    output_dir: src/generated/types/OPCUA/EUROMAP83
    extra_header: |
      #![allow(non_camel_case_types)]
      #![allow(clippy::upper_case_acronyms)]
  - type: types
    file_path: resources/OPCUA/EUROMAP/BSD/Opc.Ua.PlasticsRubber.ImmToRobot.bsd
    output_dir: src/generated/types/OPCUA/EUROMAP79
  - type: types
    file_path: resources/OPCUA/EUROMAP/BSD/Opc.Ua.PlasticsRubber.HotRunner.NodeSet2.bsd
    output_dir: src/generated/types/OPCUA/EUROMAP82_2
    extra_header: |
      #![allow(non_camel_case_types)]
      #![allow(clippy::upper_case_acronyms)]

  - type: nodes
    name: EUROMAP83
    file_path: resources/OPCUA/EUROMAP/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml
    output_dir: src/generated/nodes/OPCUA/EUROMAP83
    max_nodes_per_file: 10000
    own_namespaces:
      - "http://opcfoundation.org/UA/PlasticsRubber/GeneralTypes/"
      - "http://www.euromap.org/euromap83/"
    imported_namespaces:
      - "http://opcfoundation.org/UA/DI/"
      - "http://opcfoundation.org/UA/"
    types:
      - file_path: resources/OPCUA/EUROMAP/Opc.Ua.PlasticsRubber.IMM2MES.NodeSet2.xsd
        root_path: opcua::types
      - file_path: resources/OPCUA/EUROMAP/Opc.Ua.PlasticsRubber.ImmToRobot.xsd
        root_path: opcua::types
      - file_path: external/rust-opcua/schemas/1.05/Opc.Ua.Types.xsd
        root_path: opcua::types
```

</details>

When I try to generate code using `async-opcua-codegen codegen.yml`, I receive the following message:
```
Error: XML(XmlError { span: 218204..218229, error: MissingContent })
```

I wasn't sure how to best debug this error, so I just printed the area within the context of `async-opcua-xml::schema::ua_node_set#load_nodeset2_file`

<summary><h1>Print snippet</h1></summary>
<details>

```diff
diff --git a/async-opcua-xml/src/schema/ua_node_set.rs b/async-opcua-xml/src/schema/ua_node_set.rs
index f9373a6a..197b550a 100644
--- a/async-opcua-xml/src/schema/ua_node_set.rs
+++ b/async-opcua-xml/src/schema/ua_node_set.rs
@@ -28,10 +28,30 @@ pub struct NodeSet2 {

 /// Load a NodeSet2 file from an XML file. `document` is the content of a NodeSet2.xml file.
 pub fn load_nodeset2_file(document: &str) -> Result<NodeSet2, XmlError> {
-    let document = Document::parse(document).map_err(|e| XmlError {
-        span: 0..1,
-        error: crate::error::XmlErrorInner::Xml(e),
-    })?;
+    let head = &document[218204..218229];
+    println!("Attempting to read document: {:#?}", &head);
```

The above then returns this message for EUROMAP83
```
Attempting to read document: "<uax:String></uax:String>"
```
___
# Potentially related issue from original repository surrounding EUROMAP NodeSet generation
https://github.com/locka99/opcua/issues/46

einarmo, 2025-01-22:
I understand why this happens. I think we can relax the way we parse values using `FromValue` in opcua-xml.

Does #31 fix your issue?

evanjs, 2025-01-22:
Thank you for the quick response!
Before I try to dig into it too much, let me share what happens when testing PR 31:

<details>
<summary><h1><code>XmlError: NoRootNode</code><h1></summary>

```
Running data type code generation for resources/OPCUA/EUROMAP/BSD/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.bsd
Loading types from resources/OPCUA/EUROMAP/BSD/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.bsd
Found 28 raw elements in the type dictionary.
Generated code for 28 types
Writing 28 types to src/generated/types/OPCUA/EUROMAP83
Running data type code generation for resources/OPCUA/EUROMAP/BSD/Opc.Ua.PlasticsRubber.ImmToRobot.bsd
Loading types from resources/OPCUA/EUROMAP/BSD/Opc.Ua.PlasticsRubber.ImmToRobot.bsd
Found 1 raw elements in the type dictionary.
Generated code for 1 types
Writing 1 types to src/generated/types/OPCUA/EUROMAP79
Running data type code generation for resources/OPCUA/EUROMAP/BSD/Opc.Ua.PlasticsRubber.HotRunner.NodeSet2.bsd
Loading types from resources/OPCUA/EUROMAP/BSD/Opc.Ua.PlasticsRubber.HotRunner.NodeSet2.bsd
Found 3 raw elements in the type dictionary.
Generated code for 3 types
Writing 3 types to src/generated/types/OPCUA/EUROMAP82_2
Running node set code generation for resources/OPCUA/EUROMAP/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml
Loading node set from resources/OPCUA/EUROMAP/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml
Found 961 nodes in node set
Error: XML(XmlError { span: 0..1, error: Xml(NoRootNode) })
```

</details>

Happy to share more information as needed

einarmo, 2025-01-22:
That's a little weird. The error you're getting there is from the library we use to load the XML file to begin with. The logging is bad, and something we should improve. I _think_ this error comes from loading one of the `types` XSD files you added as dependencies to the EUROMAP83 codegen. Could you double check that they are all valid XML files?

evanjs, 2025-01-22:
No need!

Your earlier reply set me on the right track

When I was creating a repository for a minimum reproducible example, I realized that there was some weirdness in the EUROMAP XML, XSD, BSD files I had persisted in my project.

Some were different from upstream, and some were (empty) placeholders for files that don't exist (e.g. I'd have an empty XSD file when the associated standard didn't provide one)

After ensuring the validity of my resources (or at least, everything matches what is provided by EUROMAP), the generation seems to complete successfully.

I did go back to verify what happens if I try to generate without the aforementioned PR, and it indeed fails, as expected.

___

TL;DR
- The PR _**does**_ fix the issue raised in my original post
- The subsequent issue I experienced (`NoRootNode`) was due to bad XML and related files, and can be disregarded.

Thank you!

einarmo, 2025-01-22:
Fantastic. I'll close this issue once #31 is merged then, and I'll see about adding some context to errors so that at least the logs contain _which_ file failed and how.

Feel free to reach out if you run into other issues or limitations. Custom codegen, while something we want to support, is definitely a bit limited still.

evanjs, 2025-01-22:
> ... Custom codegen, while something we want to support, is definitely a bit limited still.

Might that be the reason I encounter this type of error?
```
Error: Other("Unsupported content type of type DataSetMetaDataType")
```

einarmo, 2025-01-23:
Yup. This is probably more a limitation, seems we hit this if we get a `ComplexType` node in an xsd file that contains `simpleContent` or a `complexContent` with an `Extension`. To be honest I'm not entirely sure what that means in this case. It could be that supporting this is easy, but it would probably require a little extra work.

https://github.com/FreeOpcUa/async-opcua/blob/9790a36c522c1e7f5db40078bd7fc943e7eb4fdd/async-opcua-codegen/src/nodeset/value.rs#L717, if you're interested in figuring it out. It's often not really obvious how to interpret OPC-UA XSD schemas. The Value logic in the codegen in particular is _really_ complicated.

I might be able to help, if I find the time, which XSD file is this?

evanjs, 2025-01-23:
Figured I should get the reproduction repository up and share it here before I try investigating
As well as provide as much information as I can, as I'm not certain which file is causing issues here
Here is the repository: https://github.com/evanjs/opcua-async-rust-euromap-codegen-repro

The repository has a helper script to retrieve the relevant EUROMAP specifications

___ 
To answer your question as best as I can:
`DataSetMetaDataType` seems to be a complex type from the core Opc.Ua.Types schema:
https://github.com/OPCFoundation/UA-Nodeset/blob/dee199135d85a5cb0dc5db25e27228b1f9d7ea0d/Schema/Opc.Ua.Types.xsd#L1178-L1190

I _believe_ I am encountering this error when attempting to generate nodes for EUROMAP 83
However, I am not certain which usage is causing the issue

evanjs, 2025-01-23:
@einarmo weird... I must not have grabbed your PR properly at first...

```
Error: CodeGenError { kind: Other("Unsupported content type of type DataSetMetaDataType"),
context: Some("generating node ns=1;i=6104"),
file: Some("EUROMAP/EUROMAP_79/Opc.Ua.PlasticsRubber.ImmToRobot.xml") }
```

IIUC, this resolves to `1:OPC40079ImmFixedDataSet`, as I initially somewhat expected

einarmo, 2025-01-23:
Huh. Yeah, we should really support those, but we don't. I can probably fix this. Will have a look over the next few days.

einarmo, 2025-01-23:
Actually, I think it's a bug. The comment on the bit I linked above states that we should be looking at `Extension`, but we target `Restriction`, which I don't see any instance of in the code at all... 

Could you try df25b7822156e8bc041f311e145be835e45fecf7 ?

evanjs, 2025-01-23:
Seriously appreciate the prompt feedback, thank you!

```
Error: CodeGenError {
	kind: Other("ListOf type ListOfEnumDescription is not supported, use ListOfExtensionObject instead"),
	context: Some("generating node ns=1;i=6104"),
	file: Some("EUROMAP/EUROMAP_79/Opc.Ua.PlasticsRubber.ImmToRobot.xml")
}
```

einarmo, 2025-01-23:
Oh, hm, I think I get why this is happening as well. I will need to do a little work to add support for this, but these are types defined in the core namespace, so we really should support them.

I won't be quite as quick with this one, I think ;)

einarmo, 2025-01-23:
So I went and looked at your repro and managed to get it to generate, then fixed a couple minor issues that popped up when I tried to compile it. Codegen should work with e20562189323b07097c6264323739f8a69084307.

You do have to change a few things in your config:

```diff
diff --git a/codegen.yml b/codegen.yml
index bb554d0..b855940 100644
--- a/codegen.yml
+++ b/codegen.yml
@@ -10,9 +10,6 @@ targets:
     output_dir: src/generated/opcua/types/euromap83
     enums_single_file: true
     structs_single_file: true
-    extra_header: |
-      #[allow(unused)]
-      mod opcua { pub use crate as types; }
 
   # EUROMAP 79 Types
   - type: types
@@ -57,4 +54,4 @@ targets:
       - file_path: external/async-opcua/schemas/1.05/Opc.Ua.Types.xsd
         root_path: opcua::types
       - file_path: EUROMAP/EUROMAP_83/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xsd
-        root_path: opcua::types
+        root_path: crate::generated::opcua::types::euromap83
```

I also added `mod` files as appropriate, of course.

It still won't quite compile, there are a few things missing:

 - You need a default impl for a few enums. You can just write those yourself, easy.
 - You need an `ObjectId` and `DataTypeId` enum. We generate those from CSVs. You maybe don't have any node ID CSVs? I do think we should look into writing a generator that uses `NodeSet2`, but it's a bit of work. You may have to find a workaround.
 - You may get some warnings. You can probably silence those with `extra_header` or by just disabling them for the entire generated module. None of them are particularly bad, just a couple of naming convention things and some unnecessary imports.

evanjs, 2025-01-23:
@einarmo thank you!
Messing around with it now, but the CSV files should be there
```
❯ fd . EUROMAP -e csv --no-ignore
EUROMAP\EUROMAP_77\NodeIds.csv
EUROMAP\EUROMAP_79\Opc.Ua.PlasticsRubber.ImmToRobot.csv
EUROMAP\EUROMAP_82_2\NodeIds.csv
EUROMAP\EUROMAP_83\NodeIds.csv
```

einarmo, 2025-01-23:
Then you should have everything you need, just need to add an `ids` target to the codegen config file, and re-export those at the root of your crate.

einarmo, 2025-01-24:
Closing this now that #34 is merged. Feel free to open another issue if you run into more trouble, but I think the codegen side of things should be resolved.

closed 2025-01-24 by einarmo