FreeOpcUa/async-opcua #54: Write lock occurred in .set_value (used in .add_write_callback)

AndrDm, 2025-03-04:
I've got an issue with writing variables from a client to an async-opcua server (maybe this is not an issue but incorrect usage from my side).
In short words: on server side I have a callback like this:
```rust
    manager
        .inner()
        .add_write_callback(v1_node.clone(),  move |v, _| {
            if let Some(Variant::Int32(val)) = v.value {
                println!("New Value: {}", val);
                mgr_ref // Issue is write lock in .set_value:
                    .set_value(&subscriptions, &v1_node, None, v)
                    .unwrap();
                println!("Write callback - end");
                StatusCode::Good
            } else {
                StatusCode::BadTypeMismatch
            }
        });
```
and any attempt to write v1 from client (using UA Expert, for example) caused "freezing", which is write lock (of address_space) in [macro_rules! trace_write_lock] (https://github.com/FreeOpcUa/async-opcua/blob/aca950ae6ba080ee44528c929306bf8aefec1548/async-opcua-core/src/lib.rs#L108).
The call **address_space.force_unlock_write();** before .set_value will help for sure, but this is not normal.

However, if I remove the callback entirely, attempts to write to the variable will cause **BadNotWritable** error.
But the variable was created as writable, of course (I can open separate issue if needed):

```rust
    VariableBuilder::new(&v1_node, "v1", "v1")
        .data_type(DataTypeId::Int32)
        .value(1)
        .writable() // << Writable!
        .organized_by(&sample_folder_id)
        .insert(&mut *address_spacel);
```
[Full code and screen recordings](https://github.com/AndrDm/async-opcua-LabVIEW/tree/master/samples/simple-server-issues).

Any idea what could be wrong?

By the way, the [Demo Server](https://github.com/FreeOpcUa/async-opcua/tree/master/samples/demo-server) from original samples also have the same issue — the [scalar variables](https://github.com/FreeOpcUa/async-opcua/blob/master/samples/demo-server/src/scalar.rs) created as writable, but they are not really writable and caused BadNotWritable Error.

einarmo, 2025-03-05:
I've seen this before, and perhaps we need to think about how to improve this flow. There was some discussion at one point about improving this.

Fundamentally, the issue is that write callbacks are supposed to be used with read callbacks, the idea is that the value is stored in some underlying system. For example, if you created an OPC-UA server for some control system, you would _read_ values from the control system, and _write_ values to the same system. So you would define both a read callback and a write callback, then have both read from the same underlying system.

What is missing, really, that we need to figure out, is a way to make it so that `Write` causes the value to be stored in the address space, which you _may_ want if you define only a write callback and not a read callback.

We really need to find a design for this that works well, and is reasonably intuitive.

einarmo, 2025-03-05:
I can see a few possible solutions:

 - If a read callback is not set, write to the address space. This is fairly confusing IMO, since it's not entirely obvious that this happens.
 - Always write to the address space. This works, but is potentially expensive, especially if the values written are large.
 - Have an option on the variable to allow writing (Writable is an OPC-UA concept and separate from this, I think), this could even be more general, so that we allow writing to non-value properties of nodes.
 - Change the write callback to take a reference to the address-space, so that writers can do that themselves. This will need to be a breaking change (unless we do some _real_ magic).

oroulet, 2025-03-05:
Is that issue linked to https://github.com/FreeOpcUa/async-opcua/pull/15/commits/9b983aaf39f6f72f99d1f218e99fcaebd008501a  ?

I propose to merge that commit or a version of it so writting works by default. Then we just make it very clear in docstring what is happening. That struct is already called SImpleNodeManager or similar so it already warns about limitatiois...

One could also have another NodeManager requiring to set up write callbacks on an underlying system.

btw logging is lagging here, if we require a callback to be defined, we should log an error

einarmo, 2025-03-05:
That's in your open PR, right @oroulet? I'll follow up on it and we can get it merged. That would fix your issue @AndrDm, by implementing the second option. As we discussed in that PR, since it's just the SimpleNodeManager we're fine with adding some simplified features. Advanced users can always pick a more generic node manager implementation.

einarmo, 2025-03-05:
It's fine, I'll look into getting your PR through. If you don't mind, I can polish it up if needed.

AndrDm, 2025-03-06:
Thank you so much, guys, got this! Unfortunately, I can't contribute very actively because, to be honest, my Rust skills are not strong enough yet (I'm a beginner and using this project for learning). However, I can test this with different OPC UA Clients as well as PLCs (would appreciate a small example of how to implement write callback properly using write to the address space). Once read/write is working as expected, I'll proceed to the method calls and extension objects.

einarmo, 2025-03-06:
The way it will work once the PR is merged, if you define a write callback, data will _not_ be written to the address space, but if you _don't_ have a write callback, it will.

Typically, this means you should either define both a `write` and a `read` callback, only a `read` callback, or neither.

If you need behavior beyond this I think you will need to use something other than the `SimpleNodeManager`, like extending the `CustomNodeManager` directly. See the implementation of `SimpleNodeManager` for how you can do this: https://github.com/FreeOpcUa/async-opcua/blob/master/async-opcua-server/src/node_manager/memory/simple.rs, at least for the time being.

einarmo, 2025-03-07:
@AndrDm this change is now merged to master. The way it works now is that if you want values to be stored in the address space, you need to _not_ define a write callback at all, just set `writeable` as you already do.

AndrDm, 2025-03-10:
> [@AndrDm](https://github.com/AndrDm) this change is now merged to master. The way it works now is that if you want values to be stored in the address space, you need to _not_ define a write callback at all, just set `writeable` as you already do.

Yep, it seems to be okay now. Thank you again!

closed 2025-03-10 by AndrDm