FreeOpcUa/async-opcua #63: API for creating new session from endpoint is private

irfanghat, 2025-03-15:
Hi, I was following the documentation in an attempt to create a simple client. However, it seems as though there's no public API for `new_session_from_endpoint(endpoint, IdentityToken::Anonymous)`

[image]

einarmo, 2025-03-16:
The docs are wrong. We renamed some of these methods. I don't believe that specific method even exists anymore on the latest commit on master.

I'll update the docs. I think you want `connect_to_matching_endpoint`, or `connect_to_endpoint_directly`, depending on whether or not you want to provide an _exact_ endpoint, or just something close enough.

irfanghat, 2025-03-16:
Ok, thanks for the response @einarmo !

einarmo, 2025-03-16:
We'll cut a new release soon-ish I think, there are some details to be finished up. I've updated the docs, so I'm closing this issue now.

closed 2025-03-16 by einarmo