FreeOpcUa/async-opcua #73: linking fails for fuzz

AiyionPrime, 2025-04-03:
This might be an archlinux issue, just as heads up.

```console
[...]
  Downloaded rustix v0.38.43
  Downloaded jiff v0.2.4
  Downloaded tokio v1.43.0
  Downloaded ring v0.17.8
  Downloaded 109 crates (11.2 MB) in 1.39s (largest was `ring` at 4.2 MB)
   Compiling async-opcua-fuzz v0.0.0 (/home/aiyion/Documents/async-opcua/fuzz)
error: linking with `cc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/aiyion/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/aiyion/.cargo/bin:/opt/clang-format-static:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/opt/android-sdk/platform-tools:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/aiyion/.local/bin:/home/aiyion/bin:/home/aiyion/go/bin:/opt/uaexpert/bin:~/.dotnet/tools" VSLANG="1033" "cc" "-m64" "/tmp/rustc9UvOX0/symbols.o" "<2 object files omitted>" "-Wl,--as-needed" "-Wl,-Bstatic" "/home/aiyion/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/{libstd-1e40119a663b1e41.rlib,libpanic_unwind-c17245bd091354df.rlib,libobject-cd96b354c08fae15.rlib,libmemchr-65e636254baf89ab.rlib,libaddr2line-10dcc9a2e78b2b1e.rlib,libgimli-73aed197564ca365.rlib,librustc_demangle-6d92069f05504d6a.rlib,libstd_detect-3ddba380cdb712f3.rlib,libhashbrown-1c9b748326f090af.rlib,librustc_std_workspace_alloc-61109fcaa2782b03.rlib,libminiz_oxide-4f430c1ad3399055.rlib,libadler-396e0f8ad43d4281.rlib,libunwind-b3ec995576ac8387.rlib,libcfg_if-7a8d556bd2496467.rlib,liblibc-2cd05d129739cd79.rlib,liballoc-dcbc78300c47b651.rlib,librustc_std_workspace_core-3d3fdd80ec3c369e.rlib,libcore-919178916a309d3e.rlib,libcompiler_builtins-6497188c539eb018.rlib}" "-Wl,-Bdynamic" "-lgcc_s" "-lutil" "-lrt" "-lpthread" "-lm" "-ldl" "-lc" "-Wl,--eh-frame-hdr" "-Wl,-z,noexecstack" "-L" "/home/aiyion/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-o" "/home/aiyion/Documents/async-opcua/target/debug/deps/fuzz_comms-49d85d80749c4f7f" "-Wl,--gc-sections" "-pie" "-Wl,-z,relro,-z,now" "-nodefaultlibs"
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: /usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/14.2.1/../../../../lib/Scrt1.o: in function `_start':
          (.text+0x1b): undefined reference to `main'
          collect2: error: ld returned 1 exit status
          
  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#rustc-link-lib)

error: linking with `cc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/aiyion/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/aiyion/.cargo/bin:/opt/clang-format-static:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/opt/android-sdk/platform-tools:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/aiyion/.local/bin:/home/aiyion/bin:/home/aiyion/go/bin:/opt/uaexpert/bin:~/.dotnet/tools" VSLANG="1033" "cc" "-m64" "/tmp/rustcFi04Hh/symbols.o" "<2 object files omitted>" "-Wl,--as-needed" "-Wl,-Bstatic" "/home/aiyion/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/{libstd-1e40119a663b1e41.rlib,libpanic_unwind-c17245bd091354df.rlib,libobject-cd96b354c08fae15.rlib,libmemchr-65e636254baf89ab.rlib,libaddr2line-10dcc9a2e78b2b1e.rlib,libgimli-73aed197564ca365.rlib,librustc_demangle-6d92069f05504d6a.rlib,libstd_detect-3ddba380cdb712f3.rlib,libhashbrown-1c9b748326f090af.rlib,librustc_std_workspace_alloc-61109fcaa2782b03.rlib,libminiz_oxide-4f430c1ad3399055.rlib,libadler-396e0f8ad43d4281.rlib,libunwind-b3ec995576ac8387.rlib,libcfg_if-7a8d556bd2496467.rlib,liblibc-2cd05d129739cd79.rlib,liballoc-dcbc78300c47b651.rlib,librustc_std_workspace_core-3d3fdd80ec3c369e.rlib,libcore-919178916a309d3e.rlib,libcompiler_builtins-6497188c539eb018.rlib}" "-Wl,-Bdynamic" "-lgcc_s" "-lutil" "-lrt" "-lpthread" "-lm" "-ldl" "-lc" "-Wl,--eh-frame-hdr" "-Wl,-z,noexecstack" "-L" "/home/aiyion/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-o" "/home/aiyion/Documents/async-opcua/target/debug/deps/fuzz_deserialize-cc740f7a62537c33" "-Wl,--gc-sections" "-pie" "-Wl,-z,relro,-z,now" "-nodefaultlibs"
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: /usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/14.2.1/../../../../lib/Scrt1.o: in function `_start':
          (.text+0x1b): undefined reference to `main'
          collect2: error: ld returned 1 exit status
          
  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#rustc-link-lib)

error: could not compile `async-opcua-fuzz` (bin "fuzz_comms") due to 1 previous error
warning: build failed, waiting for other jobs to finish...
error: could not compile `async-opcua-fuzz` (bin "fuzz_deserialize") due to 1 previous error

```

oroulet, 2025-04-03:
I have had that issue on ubuntu. I think it was a bug. tried master?
I really cannot remember

AiyionPrime, 2025-04-03:
This is async-opcua on branch master from this Morning.

einarmo, 2025-04-03:
In order to make it not fail in IDEs and such, since it requires a nightly compiler, I made it require a feature `nightly`. Without that it doesn't actually define a `main`, so it fails like this.

```
cargo fuzz run fuzz_comms --features nightly
```

of course you also need the normal dependencies for `cargo-fuzz`. Are you seeing this without trying to run the fuzz target yourself?

einarmo, 2025-04-03:
You get this from `cargo build` I see. That's not ideal. I have an idea for how that can be fixed, and I'll improve the code a bit...

einarmo, 2025-04-03:
Fixed in #74. Though I don't love the solution.

einarmo, 2025-04-04:
Closed in #74

closed 2025-04-04 by einarmo