FreeOpcUa/async-opcua #99: Write callback causes OutOfRange error

andergisomon, 2025-05-08:
Crate version: 0.15.1
OS: Pop!_OS 22.04 LTS x86_64

Write callback consistently produces runtime error:

```
thread 'tokio-runtime-worker' panicked at /home/ander/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/async-opcua-server-0.15.1/src/subscriptions/monitored_item.rs:377:14:
called `Result::unwrap()` on an `Err` value: OutOfRangeError(())
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[2025-05-08T01:12:52Z ERROR opcua_server::session::controller] Request panic! task 45 panicked with message "called `Result::unwrap()` on an `Err` value: OutOfRangeError(())"
[2025-05-08T01:12:52Z ERROR opcua_server::session::controller] Unexpected error in message handler: task 45 panicked with message "called `Result::unwrap()` on an `Err` value: OutOfRangeError(())"
```

Client writes are tested with UaExpert and FUXA (via node-opcua).

My write callback snippet:

```
        let write_ar1_lights_to_shmem = move |val_from_client: DataValue, _range: &NumericRange| -> StatusCode {
            log::info!("write_ar1_lights_to_shmem entered");
        
            let file = match OpenOptions::new().read(true).write(true).open(SHM_PATH) {
                Ok(f) => f,
                Err(e) => {
                    log::error!("Failed to open shared memory file: {}", e);
                    return StatusCode::Bad;
                }
            };
        
            log::info!("shmem file opened");
            let mut mmap = map_shared_memory(&file);
            let mut data = read_data(&mmap);
            log::info!("read_data mmap success");

            match val_from_client.value {
                Some(Variant::UInt32(n)) => {
                    data.area_1_lights = n;
                    write_data(&mut mmap, data);
                    StatusCode::Good
                }
                other => {
                    log::error!("Unexpected variant: {:?}", other);
                    StatusCode::Bad
                }
            }
        };
        // Client write callback
        manager
        .inner()
        .add_write_callback(
            ar1_lights_node.clone(),
            write_ar1_lights_to_shmem
        );
```

Callback that does nothing produces the same error:

```
        let write_ar1_lights_to_shmem = move |val_from_client: DataValue, _range: &NumericRange| -> StatusCode {
            log::info!("write_ar1_lights_to_shmem entered");
            // do nothing
            return StatusCode::Good
        };
```

andergisomon, 2025-05-08:
Ok so maybe I did it wrong. Here's my write callback:

```
    {
        let address_space = address_space.read(); // acquire read lock
        let node_type = address_space.find_node(&ar1_lights_hmi_cmd_node).unwrap();

        let ar1_lights_hmi_cmd_var_val = match node_type {
            NodeType::Variable(ar1_lights_hmi_cmd_var) => {
                Some(ar1_lights_hmi_cmd_var.value(
                    TimestampsToReturn::Both,
                    &NumericRange::None,
                    &opcua::types::encoding::DataEncoding::Binary,
                    1.0,
                ))
            },
            _ => None,
        };
        
        let val = ar1_lights_hmi_cmd_var_val.expect("node not a variable");                

        // Client write callback
        let write_ar1_lights_to_shmem = move |val_from_client: DataValue, _range: &NumericRange| -> StatusCode {
            log::info!("write_ar1_lights_to_shmem entered");
        
            let file = match OpenOptions::new().read(true).write(true).open(SHM_PATH) {
                Ok(f) => f,
                Err(e) => {
                    log::error!("Failed to open shared memory file: {}", e);
                    return StatusCode::Bad;
                }
            };
        
            log::info!("shmem file opened");
            let mut mmap = map_shared_memory(&file);
            let mut data = read_data(&mmap);
            log::info!("read_data mmap success");

            match val.clone().value {
                Some(Variant::UInt32(n)) => {
                    log::info!("SERVER READS VALUE: {}", n);
                    data.area_1_lights_hmi_cmd = n;
                    write_data(&mut mmap, data);
                    log::info!("mmap write_data executed");
                    StatusCode::Good
                }
                other => {
                    log::error!("Unexpected variant: {:?}", other);
                    StatusCode::Bad
                }
            }
            // return StatusCode::Good
        };
        // Client write callback
        manager
        .inner()
        .add_write_callback(
            ar1_lights_hmi_cmd_node.clone(),
            write_ar1_lights_to_shmem
        );
```

This results in no panics, but when the client writes values to the tag, it still remains at 0. The value the server reads is always 0.

einarmo, 2025-05-08:
The `OutOfRange` thing is indeed a bug. I think it is caused if you write a value that takes the current value of the variable back in time.

Not sure what is happening beyond that, though you should be very careful about holding locks on the `address_space`.

einarmo, 2025-05-08:
Do note that if your write callback returns a `Bad` status code, the value will not be stored in the address space.

einarmo, 2025-05-08:
I don't think that's right, since in fact `TimestamptsToReturn` does nothing at all in that method.

Anyway, I know what is causing the panic, and will have a fix out in a minute.

einarmo, 2025-05-08:
This particular issue should be fixed with #100.

shanithkk, 2025-05-21:
Hi @einarmo when we can expect the release with this fix, im also facing the same issue. currently using github directly.

einarmo, 2025-07-10:
I was going to cut a release once #121 is merged, and I find time to update the changelog. We probably should have a slightly higher frequency of releases.