FreeOpcUa/async-opcua #102: Invalid OPCUA sequence numbers sent by the server

evanjs, 2025-05-09:
Note: this issue also appeared on `locka99/opcua` under https://github.com/locka99/opcua/issues/190

This can be seen when using e.g. the web UI for Ignition from Inductive Automation, as noted in the original issue.

I noticed the behavior again when migrating to this repository, and wanted to raise the issue here before I forget, make sure I'm not missing anything simple.

I did notice we've done some work on sequence numbers lately
https://github.com/FreeOpcUa/async-opcua/commit/83d12f0367e3ec250bb90f604338f5f32d641278

I did not verify the behavior before and after this commit

einarmo, 2025-05-09:
I haven't seen this behavior. Do you have data on what request it starts with? It does indeed look like a missing package.

evanjs, 2025-05-09:
Right, sorry, forgot the main detail of when this appears in the logs
Didn't realize the original issue also lacks much beyond "this happens", haha oops

I can observe this from the Ignition client when subscribed via the Ignition OPC Quick Client
I _believe_ Ignition Designer results in the same behavior as well, but the Quick Client is much easier to use/test 

___
## How to Reproduce
Assuming you have connected to an OPCUA Server using the Ignition client:

Navigate to the `OPC Quick Client` page

`Config` (Sidebar) > `OPC CLIENT` (Header/Group) > `OPC Quick Client`
URL path: `/web/config/opc.quickclient`

Interacting with a tag from the address space browser using the associated action column (`[s]`) should add the tag to the `Subscription 1` interface below

Then, `Subscription name` and `Rate (ms)` can be adjusted before the subsciption is saved using `Set` 

___
### How often does this happen?

On this page, if I subscribe to a tag, I see the sequence mismatch after... something like 4 minutes with my current settings
___
### Is it consistent?
It does appear to be consistent once triggered
```
OpcUaSubscriptionManager | 09May2025 14:13:40 | [id=1] expected sequence=6, received sequence=7. Calling Republish service...
OpcUaSubscriptionManager | 09May2025 14:09:32 | [id=1] expected sequence=5, received sequence=6. Calling Republish service...
OpcUaSubscriptionManager | 09May2025 14:05:24 | [id=1] expected sequence=4, received sequence=5. Calling Republish service...
```
___
### Did this happen before?
It was definitely more prominent with `locka99/opcua` (something like every 2 minutes after I worked around the Keepalive stuff misbehaving), but I _do_ still notice the behavior here

I can try to get a minimum reproducible example up if it seems like the issue is diffiuclt to reproduce based on this information alone

___
Potentially related settings on my OPCUA server:
```rust
server_config.limits.subscriptions.min_publishing_interval_ms = 2000.0;
server_config.limits.subscriptions.min_sampling_interval_ms = 1000.0;
```

einarmo, 2025-05-09:
I see. These are notification sequence numbers. I initially assumed they were message sequence numbers, but when those are out of sync I'm reasonably sure the client should just kill the connection.

I will see if I can reporoduce. The client you're using is proprietary, which makes it a lot harder. I know many clients don't monitor subscription sequence numbers like this.

You're not seeing anything in the server logs? Notification messages can be dropped if the internal message queue goes full. I'm realizing that we should probably assign sequence numbers later, so that if we drop a notification we don't send out of sequence notification messages.

One thing though, from reading the standard, I can't actually tell if we are _required_ to send notification messages in any kind of sequence. Sequence numbers should be unique, and incrementing, but gaps seems to be fine. I think the client is doing something wrong here. It receives notification 5, then in the next iteration it asks to re-publish 5 because it thinks the notifications are out of order? How does that help? It doesn't make any sense, at least from my reading of the standard.

einarmo, 2025-05-09:
It's also strange how you get this after several minutes. You'd expect sequence number 4/5/6 to come quite early, and sequence numbers don't roll over until they hit u32::MAX.

evanjs, 2025-05-09:
Okay, so I don't see this with e.g. `v1` on the simple server sample
I _do_ see it with `v5` (static value) a few minutes (?) after subscribing

```
OpcUaSubscriptionManager | 09May2025 14:46:17 | [id=1] expected sequence=735, received sequence=736. Calling Republish service...
```

Also: they are using Milo for the OPC stuff :D

The warning comes from here
https://github.com/eclipse-milo/milo/blob/bd150672c2cee24e7934001eecb8ca581ddc4ffc/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/subscriptions/OpcUaSubscriptionManager.java#L563

[image]

einarmo, 2025-05-09:
Hm.. In that code you linked: `Set last sequence number only if this isn't a keep-alive`, and point 6 here: https://reference.opcfoundation.org/Core/Part4/v104/docs/5.13.1.1...

We treat keep-alives like any other notification message.

Can you test something for me? Set `limits.subscriptions.max_keep_alive_count` to something like 2 or 3, the subscribe to the `v5` node only. If I'm right you should see this error very quickly.

I'll look into this more when I have time. I think this may indeed be a bug, but as usual the standard is incredibly opaque.

einarmo, 2025-05-10:
This part of the standard is truly horrific. But the reference SDK agrees with you, so this should be fixed in #103. Want to give it a try?

evanjs, 2025-05-10:
The issue _seems_ to be resolved with the linked PR, but I will keep an eye on it, just in case
Thank you!

closed 2025-05-10 by einarmo