FreeOpcUa/async-opcua #104: Opcua Clients cannot connect

DeEchteThomas, 2025-05-11:
I am having issues when connecting to a created server, this also happens with the sample servers (for example the demo-server). The server is running locally. I tried two different clients to connect to the server. The first one is UaExpert. UaExpert returns a BadMethodInvalid when it tries to connect to the QueryServer and QueryApplication. The other client I used is a python opcua-client, which gives a similar error. Is this expected behaviour or do I need to change someting in the server.conf? For the sample server I didn't change anything and tried to connect anonymously. 

I migrated from the other Opcua repo, the samples over there did work for me out of the box

DeEchteThomas, 2025-05-11:
It seems to be not working on my linux machine. It does work however on my windows machine.

einarmo, 2025-05-11:
BadMethodInvalid makes no sense, we don't send that status code in response to anything connect related... it might be that some garbage is sent on the connection, though that part should be the same as the old SDK.

DeEchteThomas, 2025-05-11:
Hmmm strange, it seems to be a Nixos specific issue, something strange might be happening during the build. The crazy part is that it runs fine, except for the client connection part. I also tried out the simple-client and simple-server in tandem and they worked. However, when trying to connect to the server from other clients it fails again. I will try to look into it and notify you when I find the source of the problem.

DeEchteThomas, 2025-05-12:
When I try to connect using the python opcua-client to the demo-server from samples, I get the following error:

```
asyncua.common.structures104 - ERROR - Failed to execute auto-generated code from UA datatype: 

@dataclass
class LldpManagementAddressType:

    '''
    LldpManagementAddressType structure autogenerated from StructureDefinition object
    '''

    data_type = ua.NodeId.from_string('''i=18954''')

    AddressSubtype: ua.UInt32 = ua.UInt32(0)
    Address: ua.String = ua.String()
    IfSubtype: ua.PublishedDataItemsType_DataSetWriterName_Placeholder_Diagnostics_Counters_StateOperationalByMethod_Active = field(default_factory=ua.PublishedDataItemsType_DataSetWriterName_Placeholder_Diagnostics_Counters_StateOperationalByMethod_Active)
    IfId: ua.UInt32 = ua.UInt32(0)
')
Traceback (most recent call last):
  File "/nix/store/jmrpghb693prwv2wfvi025xhksiw2s6a-python3.12-asyncua-1.1.5/lib/python3.12/site-packages/asyncua/common/structures104.py", line 299, in _generate_object
    exec(code, env)
  File "<string>", line 4, in <module>
  File "<string>", line 14, in LldpManagementAddressType
AttributeError: module 'asyncua.ua' has no attribute 'PublishedDataItemsType_DataSetWriterName_Placeholder_Diagnostics_Counters_StateOperationalByMethod_Active'
uaclient.mainwindow - WARNING - showing error: %s')
uawidgets.utils - ERROR - module 'asyncua.ua' has no attribute 'PublishedDataItemsType_DataSetWriterName_Placeholder_Diagnostics_Counters_StateOperationalByMethod_Active'')
Traceback (most recent call last):
  File "/nix/store/i6ibs9jbky845m29i0lphvcar5hw8ga6-python3.12-opcua-widgets-0.6.1/lib/python3.12/site-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
             ^^^^^^^^^^^^^^^^^
  File "/nix/store/kcaz6f38g1haqmnax80jcqfhfb9gd0ns-opcua-client-gui-0.8.4/lib/python3.12/site-packages/uaclient/mainwindow.py", line 354, in connect
    self.uaclient.connect(uri)
  File "/nix/store/kcaz6f38g1haqmnax80jcqfhfb9gd0ns-opcua-client-gui-0.8.4/lib/python3.12/site-packages/uaclient/uaclient.py", line 126, in connect
    self.client.load_data_type_definitions()
  File "/nix/store/jmrpghb693prwv2wfvi025xhksiw2s6a-python3.12-asyncua-1.1.5/lib/python3.12/site-packages/asyncua/sync.py", line 110, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/jmrpghb693prwv2wfvi025xhksiw2s6a-python3.12-asyncua-1.1.5/lib/python3.12/site-packages/asyncua/sync.py", line 66, in post
    return futur.result(self.timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/99hl269v1igvjbp1znfk5jcarhzgy822-python3-3.12.8/lib/python3.12/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/nix/store/99hl269v1igvjbp1znfk5jcarhzgy822-python3-3.12.8/lib/python3.12/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/nix/store/jmrpghb693prwv2wfvi025xhksiw2s6a-python3.12-asyncua-1.1.5/lib/python3.12/site-packages/asyncua/client/client.py", line 875, in load_data_type_definitions
    return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/jmrpghb693prwv2wfvi025xhksiw2s6a-python3.12-asyncua-1.1.5/lib/python3.12/site-packages/asyncua/common/structures104.py", line 482, in load_data_type_definitions
    env = await _generate_object(dts.name, dts.sdef, data_type=dts.data_type, log_fail=log_ex)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/jmrpghb693prwv2wfvi025xhksiw2s6a-python3.12-asyncua-1.1.5/lib/python3.12/site-packages/asyncua/common/structures104.py", line 299, in _generate_object
    exec(code, env)
  File "<string>", line 4, in <module>
  File "<string>", line 14, in LldpManagementAddressType
AttributeError: module 'asyncua.ua' has no attribute 'PublishedDataItemsType_DataSetWriterName_Placeholder_Diagnostics_Counters_StateOperationalByMethod_Active'
uaclient.mainwindow - WARNING - showing error: %s')
```

When I connect to the demo-server with UaExpert, I get the following errors:

```
Calling QueryApplications on URL opc.tcp://localhost:4855/
Discovery QueryApplications on opc.tcp://localhost:4855/ failed (BadMethodInvalid), try QueryServers
Calling QueryServers on URL opc.tcp://localhost:4855/
Discovery QueryServers on opc.tcp://localhost:4855/ failed (BadMethodInvalid)
```
And this is the log I get from the server when I try to connect using the UaExpert client:
```
[... 5 lines cut ...]
2025-05-12T10:38:06.418008874+02:00 DEBUG opcua_server::session::controller - Incoming request; request_id=1 request_type=OpenSecureChannel request_handle=0
2025-05-12T10:38:06.418037508+02:00 DEBUG opcua_server::session::controller - Message security mode == None
2025-05-12T10:38:06.418467542+02:00 DEBUG opcua_server::session::controll
er - Incoming request; request_id=2 request_type=CreateSession request_handle=1000001
2025-05-12T10:38:06.418536212+02:00 DEBUG opcua_server::info - find_endpoint, url = opc.tcp://localhost:4855/
2025-05-12T10:38:06.418939435+02:00 INFO opcua_server::session::manager - Created new session with ID ns=1;i=2
2025-05-12T10:38:06.419789303+02:00 DEBUG opcua_server::session::controller - Incoming request; request_id=3 request_type=ActivateSession request_handle=1000002
[... cut ...]
2025-05-12T10:38:06.432031399+02:00 INFO opcua_server::session::manager - Closed session with ID ns=1;i=2
2025-05-12T10:38:06.432708781+02:00 DEBUG opcua_server::session::controller - Incoming request; request_id=8 request_type=CloseSecureChannel request_handle=0
2025-05-12T10:38:06.433115400+02:00 INFO opcua_server::server - Connection 1 terminated
```
It is really strange. Does anything use a C library dynamically at compile time or at runtime? Because nixos keeps all those things in the /nix/store/

oroulet, 2025-05-12:
out of curiosity I tried to run the demo server from async-opcua master against opcua-client (The python qt opcua client I wrote 10 years ago that someone really should port to qt6) and it seems to work nicely. I do not get the error you are talking about. Did you modify something on the server?

[image]

oroulet, 2025-05-12:
I cannot see how the error you see with opcua-client can be related to a system dependency. Maybe you have a very old version of opcua-client? (or one using a very old opcua-asyncio lib?)

DeEchteThomas, 2025-05-12:
But it also fails for the latest UaExpert build.

DeEchteThomas, 2025-05-12:
And the tests from the other Opcua library do seem to work

DeEchteThomas, 2025-05-12:
There are also some tests failing when I run cargo test, the following tests are failing:

```rust
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::view] session:9 translate_browse_paths_to_node_ids was not supplied with any browse paths
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::view] session:9 translate_browse_paths_to_node_ids failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadTooManyOperations for the request
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::view] session:15 browse was not supplied with any nodes to browse
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::view] session:15 browse failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadTooManyOperations for the request
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::view] session:15 browse_next was not supplied with any continuation points
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::view] session:15 browse_next failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadTooManyOperations for the request
[2025-05-12T16:10:24Z ERROR opcua_server::info] Client wants to connect with a non binary transport [
        UAString {
            value: Some(
                "wrongwrong",
            ),
        },
    ]
[2025-05-12T16:10:24Z ERROR opcua_server::authenticator] Cannot authenticate "sample1", password is invalid
[2025-05-12T16:10:24Z ERROR opcua_types::encoding] BadIdentityTokenRejected: Cannot authenticate user "sample1"
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::session] activate_session failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadIdentityTokenRejected for the request
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::attributes] session:29 read() value failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadDecodingError for the request
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::method] session:33 call(), was not supplied with any methods to call
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadTooManyOperations for the request
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::node_management] session:37 add_nodes, called with no nodes to add
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::node_management] session:37 add_nodes failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadTooManyOperations for the request
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::node_management] session:39 add_references, called with no references to add
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::node_management] session:39 add_references failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadTooManyOperations for the request
[2025-05-12T16:10:24Z ERROR opcua_types::encoding] BadSecurityChecksFailed: Signature mismatch
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::session] activate_session failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadSecurityChecksFailed for the request
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::attributes] session:43 history_read() value failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadNothingToDo for the request
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::attributes] session:43 history_read() value failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadTooManyOperations for the request
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::attributes] session:47 read(), was not supplied with any nodes to read
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::attributes] session:47 read() value failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadMaxAgeInvalid for the request
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::attributes] session:47 read() value failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadTimestampsToReturnInvalid for the request
[2025-05-12T16:10:24Z ERROR opcua_client::session::services::attributes] session:47 read() value failed
[2025-05-12T16:10:24Z ERROR opcua_client::session] Received a service fault of BadTooManyOperations for the request
2025-05-12T16:10:24Z ERROR opcua_server::session::controller] Request panic! task 823 panicked with message "Something went wrong! (Error emulation)"
[2025-05-12T16:10:24Z ERROR opcua_server::session::controller] Unexpected error in message handler: task 823 panicked with message "Something went wrong! (Error emulation)"
[2025-05-12T16:10:24Z ERROR opcua_client::transport::core] Received error BadInternalError from server. Reason: task 823 panicked with message "Something went wrong! (Error emulation)"
[2025-05-12T16:10:24Z ERROR opcua_server::session::controller] Request panic! task 854 panicked with message "Something went wrong! (Error emulation)"
[2025-05-12T16:10:24Z ERROR opcua_server::session::controller] Unexpected error in message handler: task 854 panicked with message "Something went wrong! (Error emulation)"
[... cut ...] gnosticInfo { symbolic_id: None, namespace_uri: None, locale: None, localized_text: None, additional_info: None, inner_status_code: None, inner_diagnostic_info: None }, string_table: None, additional_header: ExtensionObject { body: None } } })
[2025-05-12T16:10:25Z ERROR opcua_client::session] Received a service fault of BadTooManyOperations for the request
[2025-05-12T16:10:25Z ERROR opcua_server::session::controller] Connection failed while waiting for channel to be established: BadTimeout
```

einarmo, 2025-05-12:
These are error logs, but I don't see any test failures in what you pasted here. Some tests do produce errors, of verify that we handle them.

DeEchteThomas, 2025-05-12:
Okay I am actually stupid, it was a CertificateError all along, I fixed it and now it works. Thanks for all your time and effort

closed 2025-05-12 by DeEchteThomas