FreeOpcUa/async-opcua #112: Cannot read cert from path "/builds/kurant/rust-motion/pki/own/cert.der":

oroulet, 2025-05-23:
I suddently (after update to 0.15 from ... something)  got a test failing. Any idea what can be wrong? I cannot manage to repriduce that locally. that test has both a server and client opcua

test test_state_machine ... ok
failures:
---- test_axis2_move stdout ----
2025-05-23T19:27:22.101724Z ERROR opcua_crypto::certificate_store: Failed to get cert and private key: Cannot read cert from path "/builds/kurant/rust-motion/pki/own/cert.der": Could not open cert file /builds/kurant/rust-motion/pki/own/cert.der, Cannot read pkey from path "/builds/kurant/rust-motion/pki/private/private.pem": Cannot read pkey from path "/builds/kurant/rust-motion/pki/private/private.pem"
Error: Timeout while waiting for Idle, state is nowConnecting after 5s
failures:
    test_axis2_move

oroulet, 2025-05-25:
issue disapeard... must be some race condition in where these files are saved/read..

sigmunau, 2025-06-06:
It seems the real issue here is that the library in some situation generates a new certificate even when none are required (all security is disabled). And in this situation a tracing even with level ERROR is recorded. The reason why the test failed was that the certificate generating takes some time and causing a timeout a bit later in the test, so it did not actually panic or anything like that.

einarmo, 2025-06-06:
I'm reasonably sure we only generate a certificate if the option to generate certificates is set, and even without that you simply _cannot_ run an OPC-UA server without an application certificate, whatever security options you use.