FreeOpcUa/async-opcua #117: "Namespace index {} not in address space" when accessing from libloaded module

ZJVieth, 2025-06-24:
Hi,

For a prototype, Im trying to implement a modular OpcUa server using libloading. Im currently working with three crates: 1. main crate (opcua_server), 2. common (utils) crate (opcua_common), 3. device communication module crate (dcm_test). Both the main crate and the dcm crate depend on the common crate to define intefacing traits and utils. DCMs are how I want to scale the server, allowing me to develop isolated libraries, compile them to DLLs and plug any combination of them on servers connected to different machines. Im currently using libloading to load the compiled dlls. I also previously prototyped this using the open62541 wrapper crate, which worked well up until the MethodCallbacks (specifically the obtaining of input arguments inside a libloaded library).

I defined a ServerAccess type that I can pass from the main to the DCM. When I use my access utils from the main crate, I can add nodes to the address_space without issues, but when trying the same from the DCM, I get the following error:
```error
thread '<unnamed>' panicked at <...>\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\async-opcua-server-0.15.1\src\address_space\mod.rs:219:13:
Namespace index 2 not in address space
```
According to my console output, the namespace 2 does exist tho at every point I check the namespaces (including right before the error is thrown:
```error
Address from init: 0x2f078b69898
-- {2: "urn:PrototypeServer"}
-- NS Some(2)
Address from access: 0x2f078b69898
-- {2: "urn:PrototypeServer"} // address_space.namespaces()
-- NS Some(2) // address_space.namespace_index(SERVER_NAMESPACE_URN)
Enter libloaded module
Address from access: 0x2f078b69898
-- {2: "urn:PrototypeServer"} // address_space.namespaces()
-- NS Some(2) // address_space.namespace_index(SERVER_NAMESPACE_URN)
Address from Inside Access: 0x2f078b69898
-- {2: "urn:PrototypeServer"} // address_space.namespaces()
-- NS Some(2) // address_space.namespace_index(SERVER_NAMESPACE_URN)

thread '<unnamed>' panicked at <...>\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\async-opcua-server-0.15.1\src\address_space\mod.rs:219:13:
Namespace index 2 not in address space
```

Im running on Windows 11, this is the relevant part of the config.toml for the dcm (same dependency config for common and main):
```toml
[lib]
crate-type = ["dylib"]

[dependencies.async-opcua]
version = "0.15.1"
features = ["server"]
default-features = false
```

Im still new to Rust, so Im not sure if my approach to this is actually feasible/advisable. Sorry for the code-dump. Just wondering if I am doing something terribly wrong or there is actually a bug with the address space. Please let me know if you need any additional information.

opcua_server/src/main.rs
```rust

#[tokio::main]
async fn main() -> Result<(), GlobalServerError> {

    let server = server::init("./server.conf".into())?;

    let _ = server_util::access(
        &SERVER,
        |address_space, ns, server_api| {

            let mut address_space = address_space.write();

            let line_id = NodeId::new(ns, "line");
            ObjectBuilder::new(&line_id, "Line", "Line")
                .event_notifier(EventNotifier::SUBSCRIBE_TO_EVENTS)
                .organized_by(ObjectId::ObjectsFolder)
                .insert(address_space.deref_mut());

            let object_id = NodeId::new(ns, "main-object");
            ObjectBuilder::new(&object_id, "MainObject", "Main Object")
                .event_notifier(EventNotifier::SUBSCRIBE_TO_EVENTS)
                .organized_by(line_id)
                .insert(address_space.deref_mut());

            let field_id = NodeId::new(ns, "main-object-field-1");
            VariableBuilder::new(&field_id, "Var1", "Var 1")
[... 308 lines cut ...]
                println!("Address from Inside Access: {:?}", address_space.data_ptr());

                let mut address_space = address_space.write();
                println!("-- {:?} // address_space.namespaces()", address_space.namespaces());
                println!("-- NS {:?} // address_space.namespace_index(SERVER_NAMESPACE_URN)", address_space.namespace_index(SERVER_NAMESPACE_URN));

                let object_id = NodeId::new(ns, "dcm-object");
                ObjectBuilder::new(&object_id, "DCMObject", "DCM Object")
                    .event_notifier(EventNotifier::SUBSCRIBE_TO_EVENTS)
                    .organized_by(server_util::line_id(ns))
                    .insert(&mut *address_space);

                Ok(true)
            }
        ) {
            Ok(_) => println!("Executed server access in dcm."),
            Err(err) => println!("Failed server access in dcm: {:?}", err),
        }

        println!("Finish");

    }

}
```

ZJVieth, 2025-06-24:
This is the full stack trace:
```error
thread '<unnamed>' panicked at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\async-opcua-server-0.15.1\src\address_space\mod.rs:219:13:
Namespace index 2 not in address space
stack backtrace:
   0: std::panicking::begin_panic_handler
             at /rustc/17067e9ac6d7ecb70e50f92c1944e545188d2359/library\std\src\panicking.rs:697
   1: core::panicking::panic_fmt
             at /rustc/17067e9ac6d7ecb70e50f92c1944e545188d2359/library\core\src\panicking.rs:75
   2: opcua_server::address_space::AddressSpace::assert_namespace
             at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\async-opcua-server-0.15.1\src\address_space\mod.rs:219
   3: opcua_server::address_space::impl$1::insert<opcua_nodes::object::Object>
             at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\async-opcua-server-0.15.1\src\address_space\mod.rs:494
   4: opcua_nodes::object::ObjectBuilder::insert<opcua_server::address_space::AddressSpace>
             at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\async-opcua-nodes-0.15.1\src\lib.rs:210
   5: dcm_test::impl$1::add_test_nodes::closure$0
             at C:\Development\Projects\Linevision\Repos\prototype.linevision.architecture\linevision.module.test\DCM\dcm_test\src\lib.rs:55
   6: core::ops::function::FnOnce::call_once<dcm_test::impl$1::add_test_nodes::closure_env$0,tuple$<ref$<alloc::sync::Arc<lock_api::rwlock::RwLock<parking_lot::raw_rwlock::RawRwLock,opcua_server::address_space::AddressSpace>,alloc::alloc::Global> >,u16,ref$<allo
             at C:\Users\zino.vieth_yamadadob\.rustup\toolchains\stable-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\ops\function.rs:250
   7: common_opcua::server_util::access
             at C:\Development\Projects\Linevision\Repos\prototype.linevision.architecture\linevision.common\Opcua\common_opcua\src\server_util.rs:79
   8: dcm_test::DCMTest::add_test_nodes
             at C:\Development\Projects\Linevision\Repos\prototype.linevision.architecture\linevision.module.test\DCM\dcm_test\src\lib.rs:44
   9: dcm_test::impl$0::init
             at C:\Development\Projects\Linevision\Repos\prototype.linevision.architecture\linevision.module.test\DCM\dcm_test\src\lib.rs:32
  10: LineVisionOpcuaServer::meta::dcms::load_dcm
             at .\src\meta\dcms.rs:64
  11: LineVisionOpcuaServer::meta::dcms::load_and_init_dcms::closure$0
             at .\src\meta\dcms.rs:31
  12: scan_dir::ScanDir::read<ref$<str$>,tuple$<>,LineVisionOpcuaServer::meta::dcms::load_and_init_dcms::closure_env$0>
             at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\scan_dir-0.3.3\src\lib.rs:274
  13: LineVisionOpcuaServer::meta::dcms::load_and_init_dcms
             at .\src\meta\dcms.rs:25
  14: LineVisionOpcuaServer::main::async_block$0
             at .\src\main.rs:76
  15: core::future::future::impl$1::poll<alloc::boxed::Box<enum2$<LineVisionOpcuaServer::main::async_block_env$0>,alloc::alloc::Global> >
             at C:\Users\zino.vieth_yamadadob\.rustup\toolchains\stable-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\future\future.rs:124
  16: tokio::runtime::park::impl$4::block_on::closure$0<core::pin::Pin<alloc::boxed::Box<enum2$<LineVisionOpcuaServer::main::async_block_env$0>,alloc::alloc::Global> > >
             at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\tokio-1.45.1\src\runtime\park.rs:284
  17: tokio::task::coop::with_budget
             at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\tokio-1.45.1\src\task\coop\mod.rs:167
  18: tokio::task::coop::budget
             at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\tokio-1.45.1\src\task\coop\mod.rs:133
  19: tokio::runtime::park::CachedParkThread::block_on<core::pin::Pin<alloc::boxed::Box<enum2$<LineVisionOpcuaServer::main::async_block_env$0>,alloc::alloc::Global> > >
             at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\tokio-1.45.1\src\runtime\park.rs:284
  20: tokio::runtime::context::blocking::BlockingRegionGuard::block_on<core::pin::Pin<alloc::boxed::Box<enum2$<LineVisionOpcuaServer::main::async_block_env$0>,alloc::alloc::Global> > >
             at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\tokio-1.45.1\src\runtime\context\blocking.rs:66
  21: tokio::runtime::scheduler::multi_thread::impl$0::block_on::closure$0<core::pin::Pin<alloc::boxed::Box<enum2$<LineVisionOpcuaServer::main::async_block_env$0>,alloc::alloc::Global> > >
             at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\tokio-1.45.1\src\runtime\scheduler\multi_thread\mod.rs:87
  22: tokio::runtime::context::runtime::enter_runtime<tokio::runtime::scheduler::multi_thread::impl$0::block_on::closure_env$0<core::pin::Pin<alloc::boxed::Box<enum2$<LineVisionOpcuaServer::main::async_block_env$0>,alloc::alloc::Global> > >,enum2$<core::result::
             at C:\Users\zino.vieth_yamadadob\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\tokio-1.45.1\src\runtime\context\runtime.rs:65
[... cut ...]
This seems to be the panic call: `\src\address_space\mod.rs:219`

On a side note, implementing the actual address_space manipulating operations as part of the ServerAPI trait in the main crate makes it possible to go around this problem, but Id generally prefer to be able to do these operations directly from the modules without having to extend the common ServerAPI for every future use case. Not sure if it would be preferable from a safety point of view though.
Thank you for taking the time to look into this.

einarmo, 2025-06-24:
I'll be honest, this makes no sense to me. As you note, the `namespaces` map contains the key `2`, as you show from your logs, and the check that fails here is literally just `if !self.namespaces.contains_key(&node_id.namespace) { panic!(...) }`.

It's a little hard to debug just by looking at the code. I can try to look into it, but I doubt I'll be able to reproduce.

There's probably something obvious we're both missing. `namespaces` really isn't allowed to change, it's behind a write-lock.

einarmo, 2025-06-24:
Maybe there's some form of static randomness in `hashbrown::HashMap` that doesn't work cross dynamic rate boundary like this?

einarmo, 2025-06-25:
Right. So I'm reasonably sure this is the problem:

 - We use `hashbrown` in a few places for performance, including here.
 - `hashbrown` uses `foldhash` for hashing by default, using `RandomState` from that library, which has a _static_ shared random seed, again for performance.
 - Since the random seed is static, it is part of the _static_ part of the binary, meaning that each dynamic library has a different copy.
 - This means that you get different hashes in different dynamic libraries.

This just happens to be the first place you hit this issue, you would have hit it sooner or later, and in general is something you run into if you dynamically link libraries.

I think changing the behavior on our end is not trivial, but theoretically doable though I'm not sure it will actually fix _all_ your issues. We could continue to use a static hasher, but allow it to be manually initialized, i.e. I haven't seen that pattern before. It could be done, but I somewhat hesitate, because I'm not sure if we want to commit to supporting something like this. I'm unsure of what the actual consequences are. We use tokio, which has lots of static stuff for timers and the like, you may run into similar issues there, or in any number of other libraries that have static variables somewhere.

einarmo, 2025-06-25:
If you only needed to fix _this_ specific problem, allowing extension libraries to call `insert` using the server address space, I guess you could implement `NodeInsertTarget` for some wrapper type ensuring that all actual insertions happen in the same static context, but that wouldn't work if you needed to do anything else.

ZJVieth, 2026-02-17:
Just to wrap this up, the solution I eventually implemented was to provide a library with a server api trait, which is only implemented by the main server application, and the resulting object is then shared with the libloaded plugins to get access to the address space.

closed 2026-02-17 by ZJVieth