FreeOpcUa/async-opcua #124: The Limit max_array_length isn't respected

lov3b, 2025-07-14:
The `Limit::max_array_length` field isn't respected when decoding elements. 

I've set the `Limit::max_array_length` to `15_000`. Still, I get the following error message:

```
[2025-07-14 12:53:54] ERROR: BadDecodingError: Array length 1306 exceeds decoding limit 1000
```

My suspicion is that `DecodingOptions::default()` is used in a place where we should get the value from the `Limit` supplied to the `Server`. The issue goes away if I override the library and manually set the following constant `MAX_ARRAY_LENGTH` to 15_000. 

https://github.com/FreeOpcUa/async-opcua/blob/94a63352bdeee1fa86938e3d69278086fd483590/async-opcua-types/src/lib.rs#L33

einarmo, 2025-07-15:
Ah, that looks like the client. Did you adjust the config on the client as well?

lov3b, 2025-07-15:
Thanks, sorry for bothering you. As you pointed out the problem was with the client. We embed a small client which is used to let some clients to the server subscribe for changes. The following snippet solved it

```rust

let mut client = ClientBuilder::new()
        .max_array_length(15_000) // this line
```

closed 2025-07-15 by lov3b