FreeOpcUa/async-opcua #128: BadNonceInvalid

demnoc, 2025-07-22:
Hi! I have a problem about StatusCode `BadNonceInvalid`. The opcua server supports SecurityPolicy::Basic128Rsa15 and MessageSecurityMode::Sign. But I received this StatusCode BadNonceInvalid. I used the self-sign certifiate. Below is the trace:

> 2025-07-22T09:22:23.410199Z DEBUG Connect
> 2025-07-22T09:22:23.411059Z DEBUG Security policy = None
> 2025-07-22T09:22:23.411080Z DEBUG Security mode = None
> 2025-07-22T09:22:23.411153Z DEBUG Connecting to 10.106.251.216:7778 with url opc.tcp://10.106.251.216:7778
> 2025-07-22T09:22:23.417576Z TRACE Send hello message: HelloMessage { message_header: MessageHeader { message_type: Hello, message_size: 61 }, protocol_version: 0, receive_buffer_size: 65535, send_buffer_size: 65535, max_message_size: 327675, max_chunk_count: 5, endpoint_url: UAString { value: Some("opc.tcp://10.106.251.216:7778") } }
> 2025-07-22T09:22:23.424139Z TRACE Received acknowledgement: AcknowledgeMessage { message_header: MessageHeader { message_type: Acknowledge, message_size: 28 }, protocol_version: 0, receive_buffer_size: 65535, send_buffer_size: 65535, max_message_size: 2097152, max_chunk_count: 64 }
> 2025-07-22T09:22:23.424240Z TRACE issue_or_renew_secure_channel(Issue)
> 2025-07-22T09:22:23.424260Z DEBUG Making secure channel request
> 2025-07-22T09:22:23.424265Z DEBUG security_mode = None
> 2025-07-22T09:22:23.424271Z DEBUG security_policy = None
> 2025-07-22T09:22:23.424374Z TRACE Writing request to buffer
> 2025-07-22T09:22:23.424408Z TRACE AsymmetricSecurityHeader security policy none
> 2025-07-22T09:22:23.424418Z TRACE AsymmetricSecurityHeader = AsymmetricSecurityHeader { security_policy_uri: UAString { value: Some("http://opcfoundation.org/UA/SecurityPolicy#None") }, sender_certificate: ByteString { value: None }, receiver_certificate_thumbprint: ByteString { value: None } }
> 2025-07-22T09:22:23.424436Z TRACE No certificate / public key was supplied in the asymmetric security header
> 2025-07-22T09:22:23.424467Z TRACE AsymmetricSecurityHeader security policy none
> 2025-07-22T09:22:23.424471Z TRACE AsymmetricSecurityHeader = AsymmetricSecurityHeader { security_policy_uri: UAString { value: Some("http://opcfoundation.org/UA/SecurityPolicy#None") }, sender_certificate: ByteString { value: None }, receiver_certificate_thumbprint: ByteString { value: None } }
> 2025-07-22T09:22:23.424476Z TRACE Creating a chunk with a size of 132, data excluding padding & signature
> 2025-07-22T09:22:23.429382Z TRACE Decrypting OpenSecureChannel
> 2025-07-22T09:22:23.429437Z TRACE receive chunk final 1001:1. Length 56
> 2025-07-22T09:22:23.429462Z TRACE Received chunk with sequence number 1
> 2025-07-22T09:22:23.429572Z DEBUG Setting transport's security token
> 2025-07-22T09:22:23.429675Z TRACE Writing request to buffer
> 2025-07-22T09:22:23.429700Z TRACE Creating a chunk with a size of 98, data excluding padding & signature
> 2025-07-22T09:22:23.444734Z TRACE receive chunk final 1002:2. Length 3894
> 2025-07-22T09:22:23.444797Z TRACE Received chunk with sequence number 2
> 2025-07-22T09:22:23.444985Z DEBUG Writer is about to send a CloseSecureChannelRequest which means it should close in a moment
> 2025-07-22T09:22:23.444993Z TRACE Writing request to buffer
> 2025-07-22T09:22:23.445011Z TRACE Creating a chunk with a size of 57, data excluding padding & signature
> 2025-07-22T09:22:23.445114Z DEBUG Writer is setting the connection state to finished(good)
> 2025-07-22T09:22:23.446394Z  INFO event_loop.next: Ok(BeginConnect)
> 2025-07-22T09:22:23.459979Z DEBUG Connect
> 2025-07-22T09:22:23.461397Z DEBUG Security policy = Basic128Rsa15
> 2025-07-22T09:22:23.461426Z DEBUG Security mode = Sign
> 2025-07-22T09:22:23.461470Z DEBUG Connecting to 10.106.251.216:7778 with url opc.tcp://10.106.251.216:7778
> 2025-07-22T09:22:23.467685Z TRACE Send hello message: HelloMessage { message_header: MessageHeader { message_type: Hello, message_size: 61 }, protocol_version: 0, receive_buffer_size: 65535, send_buffer_size: 65535, max_message_size: 327675, max_chunk_count: 5, endpoint_url: UAString { value: Some("opc.tcp://10.106.251.216:7778") } }
> 2025-07-22T09:22:23.478944Z TRACE Received acknowledgement: AcknowledgeMessage { message_header: MessageHeader { message_type: Acknowledge, message_size: 28 }, protocol_version: 0, receive_buffer_size: 65535, send_buffer_size: 65535, max_message_size: 2097152, max_chunk_count: 64 }
> 2025-07-22T09:22:23.479025Z TRACE issue_or_renew_secure_channel(Issue)
> 2025-07-22T09:22:23.479229Z DEBUG Making secure channel request
> 2025-07-22T09:22:23.479239Z DEBUG security_mode = Sign
> 2025-07-22T09:22:23.479245Z DEBUG security_policy = Basic128Rsa15
> 2025-07-22T09:22:23.479318Z TRACE Writing request to buffer
> 2025-07-22T09:22:23.479615Z TRACE AsymmetricSecurityHeader = AsymmetricSecurityHeader { security_policy_uri: UAString { value: Some("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15") }, sender_certificate: ByteString { value: Some([48, 130, 3, 252, 48, 130, 2, 228, 160, 3, 2, 1, 2, 2, 1, 42, 48, 13, 6, 9, 42, 134, 72, 134, 247, 13, 1, 1, 11, 5, 0, 48, 71, 49, 11, 48, 9, 6, 3, 85, 4, 8, 12, 2, 67, 81, 49, 11, 48, 9, 6, 3, 85, 4, 6, 19, 2, 67, 78, 49, 13, 48, 11, 6, 3, 85, 4, 11, 12, 4, 97, 105, 111, 116, 49, 13, 48, 11, 6, 3, 85, 4, 10, 12, 4, 97, 105, 111, 116, 49, 13, 48, 11, 6, 3, 85, 4, 3, 12, 4, 97, 105, 111, 116, 48, 32, 23, 13, 50, 53, 48, 55, 49, 52, 48, 56, 53, 52, 51, 53, 90, 24, 15, 50, 49, 50, 53, 48, 54, 50, 48, 48, 56, 53, 52, 51, 53, 90, 48, 71, 49, 11, 48, 9, 6, 3, 85, 4, 8, 12, 2, 67, 81, 49, 11, 48, 9, 6, 3, 85, 4, 6, 19, 2, 67, 78, 49, 13, 48, 11, 6, 3, 85, 4, 11, 12, 4, 97, 105, 111, 116, 49, 13, 48, 11, 6, 3, 85, 4, 10, 12, 4, 97, 105, 111, 116, 49, 13, 48, 11, 6, 3, 85, 4, 3, 12, 4, 97, 105, 111, 116, 48, 130, 1, 34, 48, 13, 6, 9, 42, 134, 72, 134, 247, 13, 1, 1, 1, 5, 0, 3, 130, 1, 15, 0, 48, 130, 1, 10, 2, 130, 1, 1, 0, 192, 44, 5, 218, 87, 234, 179, 248, 242, 73, 250, 17, 200, 74, 191, 225, 250, 225, 82, 18, 2, 212, 84, 203, 161, 203, 202, 143, 21, 45, 42, 132, 105, 60, 119, 193, 131, 191, 9, 207, 109, 167, 91, 210, 85, 52, 46, 76, 134, 221, 255, 40, 75, 186, 158, 6, 231, 96, 110, 26, 94, 196, 68, 211, 75, 47, 22, 117, 220, 111, 183, 235, 11, 198, 194, 5, 220, 96, 36, 26, 16, 12, 188, 81, 174, 64, 214, 84, 108, 86, 154, 193, 235, 0, 177, 194, 136, 66, 25, 57, 237, 119, 216, 34, 171, 214, 131, 96, 216, 14, 1, 28, 42, 146, 31, 57, 238, 137, 87, 229, 30, 88, 37, 74, 9, 211, 140, 87, 249, 15, 241, 72, 151, 143, 117, 223, 107, 16, 1, 123, 237, 148, 42, 187, 218, 249, 122, 107, 80, 108, 137, 207, 224, 63, 71, 128, 132, 37, 225, 125, 125, 201, 115, 39, 143, 200, 59, 159, 4, 194, 127, 106, 217, 240, 106, 255, 246, 61, 21, 175, 219, 214, 77, 115, 110, 146, 247, 0, 32, 93, 176, 196, 22, 196, 122, 248, 52, 116, 136, 0, 190, 252, 105, 222, 9, 187, 185, 161, 72, 153, 130, 187, 169, 232, 76, 183, 192, 98, 105, 67, 68, 236, 235, 232, 192, 47, 128, 83, 56, 100, 183, 43, 208, 98, 74, 73, 104, 105, 48, 10, 171, 72, 48, 135, 202, 99, 58, 25, 216, 7, 235, 141, 115, 17, 191, 177, 2, 3, 1, 0, 1, 163, 129, 240, 48, 129, 237, 48, 29, 6, 3, 85, 29, 14, 4, 22, 4, 20, 82, 168, 248, 202, 107, 13, 242, 214, 40, 155, 43, 186, 70, 154, 84, 21, 178, 109, 190, 44, 48, 111, 6, 3, 85, 29, 35, 4, 104, 48, 102, 128, 20, 82, 168, 248, 202, 107, 13, 242, 214, 40, 155, 43, 186, 70, 154, 84, 21, 178, 109, 190, 44, 161, 75, 164, 73, 48, 71, 49, 11, 48, 9, 6, 3, 85, 4, 8, 12, 2, 67, 81, 49, 11, 48, 9, 6, 3, 85, 4, 6, 19, 2, 67, 78, 49, 13, 48, 11, 6, 3, 85, 4, 11, 12, 4, 97, 105, 111, 116, 49, 13, 48, 11, 6, 3, 85, 4, 10, 12, 4, 97, 105, 111, 116, 49, 13, 48, 11, 6, 3, 85, 4, 3, 12, 4, 97, 105, 111, 116, 130, 1, 42, 48, 12, 6, 3, 85, 29, 19, 1, 1, 255, 4, 2, 48, 0, 48, 14, 6, 3, 85, 29, 15, 1, 1, 255, 4, 4, 3, 2, 2, 244, 48, 32, 6, 3, 85, 29, 37, 1, 1, 255, 4, 22, 48, 20, 6, 8, 43, 6, 1, 5, 5, 7, 3, 2, 6, 8, 43, 6, 1, 5, 5, 7, 3, 1, 48, 27, 6, 3, 85, 29, 17, 4, 20, 48, 18, 134, 16, 117, 114, 110, 58, 83, 105, 109, 112, 108, 101, 67, 108, 105, 101, 110, 116, 48, 13, 6, 9, 42, 134, 72, 134, 247, 13, 1, 1, 11, 5, 0, 3, 130, 1, 1, 0, 92, 109, 180, 26, 194, 203, 119, 202, 68, 177, 194, 129, 106, 164, 37, 21, 117, 21, 249, 13, 191, 229, 99, 56, 195, 80, 161, 36, 200, 221, 111, 144, 167, 78, 147, 199, 201, 48, 69, 236, 233, 16, 145, 186, 107, 243, 110, 153, 124, 77, 109, 246, 66, 67, 114, 241, 41, 174, 178, 148, 117, 166, 33, 169, 157, 189, 118, 48, 37, 10, 103, 249, 140, 172, 255, 18, 91, 1, 239, 249, 4, 240, 113, 156, 1 [... cut ...]
> 2025-07-22T09:22:23.595251Z TRACE 00000032:  eb 21 ea fa db 01 03 00 00 00 00 00 00 00 ff ff ff ff 60 ea 00 00 00 00 00 c1 90 cf 63 6e cb 4d .!................`.........cn.M
> 2025-07-22T09:22:23.595258Z TRACE 00000064:  cd e4 2e 37 2c 37 2a 3e 52 6b ed 30 1c                                                          ...7,7*>Rk.0.
> 2025-07-22T09:22:23.595330Z DEBUG Writer is setting the connection state to finished(good)
> 2025-07-22T09:22:23.595432Z  WARN Failed to connect to server, status code: BadNonceInvalid
> 2025-07-22T09:22:23.595444Z  INFO event_loop.next: Err(BadNonceInvalid (2149842944))

einarmo, 2025-07-22:
What server is this? It seems the server is rejecting the CreateSession request, but it's hard to say why... Does it work with a different security mode like None or SignAndEncrypt. Just Sign is often a bit of an edge case, and not used that often in practice.

demnoc, 2025-07-23:
Thank you for reply. The server is "Prosys OPC UA Simulation Server". It supports all Security Modes(None, Sign, Sign&Encrypt) and all Security Policies(Basic128Rsa15, Basic256, Basic256Sha256, Aes128-Sha256-RsaOaep, Aes256-Sha256-RsaPss). 
Except Basic128Rsa15 (Sign, Sign&Encrypt), the others all succeed. The picture below shows the result

[image]
[image]

> The trace below is Basic128Rsa15  Sign&Encrypt. I also receive StattusCode BadNonceInvalid.
> 2025-07-23T01:03:38.767078Z  INFO Starting opc connect
> 2025-07-23T01:03:38.767203Z  WARN Endpoint config contains no endpoints
> 2025-07-23T01:03:38.769665Z  INFO Connecting to OPC UA server
> 2025-07-23T01:03:38.770944Z DEBUG Connect
> 2025-07-23T01:03:38.771783Z DEBUG Security policy = None
> 2025-07-23T01:03:38.771802Z DEBUG Security mode = None
> 2025-07-23T01:03:38.771875Z DEBUG Connecting to 127.0.0.1:53530 with url opc.tcp://127.0.0.1:53530/OPCUA/SimulationServer
> 2025-07-23T01:03:38.773533Z TRACE Send hello message: HelloMessage { message_header: MessageHeader { message_type: Hello, message_size: 80 }, protocol_version: 0, receive_buffer_size: 65535, send_buffer_size: 65535, max_message_size: 327675, max_chunk_count: 5, endpoint_url: UAString { value: Some("opc.tcp://127.0.0.1:53530/OPCUA/SimulationServer") } }
> 2025-07-23T01:03:38.802414Z TRACE Received acknowledgement: AcknowledgeMessage { message_header: MessageHeader { message_type: Acknowledge, message_size: 28 }, protocol_version: 0, receive_buffer_size: 8196, send_buffer_size: 8196, max_message_size: 4194240, max_chunk_count: 0 }
> 2025-07-23T01:03:38.802517Z TRACE issue_or_renew_secure_channel(Issue)
> 2025-07-23T01:03:38.802539Z DEBUG Making secure channel request
> 2025-07-23T01:03:38.802545Z DEBUG security_mode = None
> 2025-07-23T01:03:38.802550Z DEBUG security_policy = None
> 2025-07-23T01:03:38.802681Z TRACE Writing request to buffer
> 2025-07-23T01:03:38.802718Z TRACE AsymmetricSecurityHeader security policy none
> 2025-07-23T01:03:38.802729Z TRACE AsymmetricSecurityHeader = AsymmetricSecurityHeader { security_policy_uri: UAString { value: Some("http://opcfoundation.org/UA/SecurityPolicy#None") }, sender_certificate: ByteString { value: None }, receiver_certificate_thumbprint: ByteString { value: None } }
> 2025-07-23T01:03:38.802748Z TRACE No certificate / public key was supplied in the asymmetric security header
> 2025-07-23T01:03:38.802778Z TRACE AsymmetricSecurityHeader security policy none
> 2025-07-23T01:03:38.802782Z TRACE AsymmetricSecurityHeader = AsymmetricSecurityHeader { security_policy_uri: UAString { value: Some("http://opcfoundation.org/UA/SecurityPolicy#None") }, sender_certificate: ByteString { value: None }, receiver_certificate_thumbprint: ByteString { value: None } }
> 2025-07-23T01:03:38.802788Z TRACE Creating a chunk with a size of 132, data excluding padding & signature
> 2025-07-23T01:03:38.809178Z TRACE Decrypting OpenSecureChannel
> 2025-07-23T01:03:38.809222Z TRACE receive chunk final 1001:933. Length 56
> 2025-07-23T01:03:38.809243Z TRACE Received chunk with sequence number 933
> 2025-07-23T01:03:38.809356Z DEBUG Setting transport's security token
> 2025-07-23T01:03:38.809414Z TRACE Writing request to buffer
> 2025-07-23T01:03:38.809436Z TRACE Creating a chunk with a size of 117, data excluding padding & signature
> 2025-07-23T01:03:38.817439Z TRACE receive chunk intermediate 1002:934. Length 4096
> 2025-07-23T01:03:38.817506Z TRACE receive chunk intermediate 1002:935. Length 4096
> 2025-07-23T01:03:38.817528Z TRACE receive chunk intermediate 1002:936. Length 4096
> 2025-07-23T01:03:38.817544Z TRACE receive chunk intermediate 1002:937. Length 4096
> 2025-07-23T01:03:38.817558Z TRACE receive chunk final 1002:938. Length 1503
> 2025-07-23T01:03:38.817573Z TRACE Received chunk with sequence number 934
> 2025-07-23T01:03:38.817746Z DEBUG Writer is about to send a CloseSecureChannelRequest which means it should close in a moment
> 2025-07-23T01:03:38.817758Z TRACE Writing request to buffer
> 2025-07-23T01:03:38.817777Z TRACE Creating a chunk with a size of 57, data excluding padding & signature
> 2025-07-23T01:03:38.817858Z DEBUG Writer is setting the connection state to finished(good)
> 2025-07-23T01:03:38.819136Z  INFO event_loop.next: Ok(BeginConnect)
> 2025-07-23T01:03:38.829925Z DEBUG Connect
> 2025-07-23T01:03:38.831196Z DEBUG Security policy = Basic128Rsa15
> 2025-07-23T01:03:38.831226Z DEBUG Security mode = SignAndEncrypt
> 2025-07-23T01:03:38.831271Z DEBUG Connecting to 127.0.0.1:53530 with url opc.tcp://127.0.0.1:53530/OPCUA/SimulationServer
> 2025-07-23T01:03:38.831785Z TRACE Send hello message: HelloMessage { message_header: MessageHeader { message_type: Hello, message_size: 80 }, protocol_version: 0, receive_buffer_size: 65535, send_buffer_size: 65535, max_message_size: 327675, max_chunk_count: 5, endpoint_url: UAString { value: Some("opc.tcp://127.0.0.1:53530/OPCUA/SimulationServer") } }
> 2025-07-23T01:03:38.835004Z TRACE Received acknowledgement: AcknowledgeMessage { message_header: MessageHeader { message_type: Acknowledge, message_size: 28 }, protocol_version: 0, receive_buffer_size: 8196, send_buffer_size: 8196, max_message_size: 4194240, max_chunk_count: 0 }
> 2025-07-23T01:03:38.835086Z TRACE issue_or_renew_secure_channel(Issue)
> 2025-07-23T01:03:38.835301Z DEBUG Making secure channel request
> 2025-07-23T01:03:38.835310Z DEBUG security_mode = SignAndEncrypt
> 2025-07-23T01:03:38.835314Z DEBUG security_policy = Basic128Rsa15
[... cut ...]
> 2025-07-23T01:03:39.001197Z TRACE 00000032:  1c c9 09 12 fc 9d 19 00 d7 69 df 00 b6 e1 fd d4 bf 89 25 81 1c 20 4e 0e f0 79 83 84 78 f6 6f 19 .........i........%.. N..y..x.o.
> 2025-07-23T01:03:39.001204Z TRACE 00000064:  de 6c 0a 22 40 e8 57 7d 17 99 95 86 d8 35 9d 2c                                                 .l."@.W}.....5.,
> 2025-07-23T01:03:39.001363Z DEBUG Writer is setting the connection state to finished(good)
> 2025-07-23T01:03:39.001451Z  WARN Failed to connect to server, status code: BadNonceInvalid
> 2025-07-23T01:03:39.001466Z  INFO event_loop.next: Err(BadNonceInvalid (2149842944))

einarmo, 2025-07-23:
Indeed, we have mixed something up here. I'll fix it.

demnoc, 2025-07-23:
**Many thanks!**