FreeOpcUa/async-opcua #133: Load Custom Structs using Dynamic Type Loader when NodeId is not Numeric

porquey, 2025-08-05:
Hello,
I have been experimenting with custom dynamic structures via the example code in the samples: https://github.com/FreeOpcUa/async-opcua/blob/master/samples/custom-structures-client/src/bin/dynamic_client.rs 

In my example I have the following code to load in the dynamic structs:
```
        let type_tree = DataTypeTreeBuilder::new(|f| f.namespace <= 4)
            .values_per_read(100)
            .build(&session)
            .await?;
        println!("{:?}", type_tree);
        let loader = Arc::new(DynamicTypeLoader::new(Arc::new(type_tree)));
        session.add_type_loader(loader);
```

From my limited testing and debugging I see that it can load the types and encoders for dynamic types registered by the OPCUA server when the NodeId of the types (and perhaps the variables) are `Numeric`.
However in our CODESYS generated OPCUA Server, the custom types are created with NodeId Identifier: `String`.

The above code example prints the `type_tree`, and I can see that for the `dynamic_client.rs` example, it can populate the `struct_types`, `enum_types` and `encoding_to_data_type` fields correctly.
When I repeat the same code against the CODESYS server, it iterates through the data type nodes such as:
```
NodeId { namespace: 4, identifier: String(UAString { value: Some("|type|CODESYS Control for Linux SL.Application.extractionAPI") }) }
```
But the `type_tree` contains empty `struct_types`, `enum_types` and `encoding_to_data_type` fields.

I wonder if this is a known limitation of dynamic type loader, or is expected behaviour according to OPCUA spec, and whether CODESYS is doing something unique with their server.

At the minimum, it would be great if types that do not have a registered encoder can still return the raw binary, rather than panicking.

_Originally posted by @porquey in https://github.com/FreeOpcUa/async-opcua/discussions/132_

einarmo, 2025-08-05:
AFAIK the dynamic type loader does not concern itself with node ID type at all. Could you get some log output using something like `env_logger`?

porquey, 2025-08-05:
Ok I tried to dive deeper into the issue and here are my findings.
During `build()` of the type tree, I can see the code manages to get to this line: https://github.com/FreeOpcUa/async-opcua/blob/master/async-opcua-client/src/custom_types/mod.rs#L138

In this function it invokes `id.as_data_type_id()` which returns an `Err(NodeIdError)`. 
Looking into the implementation of `as_data_type_id`: https://github.com/FreeOpcUa/async-opcua/blob/61672086c2b5ba622386e6ba94503671f27fa301/async-opcua-types/src/node_id.rs#L777-L784

Here it tries to match the node identifier to `Numeric(id)`, but in our case it is `String(UAString(s))`. I'm not so familiar with the OPCUA protocol, and am not sure what data_type_id it should return in this case, or whether there is another way to infer whether this data_type is a struct, primitive or enum (perhaps based on VariantType?).

einarmo, 2025-08-05:
Hm. I don't think that is it. What that code does is recursively iterate up through the data types to find which built-in type they descend from in the type hierarchy. Note that it does `if let Ok(t) = id.as_data_type_id()`, so it handles the error returned from `as_data_type_id` locally.

It'll return an error for your type, sure, but then it will keep going

```rust
let parent = self.parent_ids.get(id)?;
self.get_data_type_variant(parent)
```

recursively traversing through its parent.

porquey, 2025-08-05:
I see, thanks for explaining. Let me do a bit more digging around this type_tree generation and then get back to you.

einarmo, 2025-08-05:
Is it a structure? If yes, then indeed, that is the problem. The dynamic type loader is meant to deserialize structures by reading their definition from OPC-UA. This doesn't appear to _have_ a type definition, so the dynamic type loader can't handle it.

einarmo, 2025-08-05:
If what you really need is just a fallback, then I do have some ideas for how to do that. It's something that I've planned to implement. Of course, that wouldn't actually let you read these structures in a meaningful way, you'd just get a binary blob.

porquey, 2025-08-06:
I see, so with the Codegen can do what Python lib is doing, but not with String NodeIds yet?

Can you elaborate on why I couldn't use the StaticTypeLoader shortcut?

einarmo, 2025-08-06:
The `StaticTypeLoader` shortcut is the part that requires numeric node IDs, though we could probably change it to be generic.

There's nothing _really_ preventing us from supporting non-numeric node IDs with codegen, but it would take a little work to implement.

porquey, 2025-08-06:
Thanks for the continuous prompt support.
Going back to the topic on whether a client could dynamically load the structs from information in the OPCUA Server, even if DataTypeDefinition field is None. I want to dig deeper on how other clients/applications are still able to do it, despite this being missing.

When connecting to our server via UAExpert, I inspected the following nodes (which have similar String NodeId) for custom struct in question:
- Variable node (|var|)
  - NodeClass: `Variable`
  - DataType: `Application.CommandRequest` (custom struct)
  - Value: Fully decoded instance of custom struct
  - References:
    - HasLocalVar: fields of struct
    - HasTypeDefinition: `BaseVariableType`
- Type node (|type|)
  - NodeClass: `DataType`
  - DataTypeDefinition: `BadAttributeIdInvalid()`
  - References:
    - HasEncoding: `Default Binary`
- Encoding node (|enc|)
  - NodeClass: `Object`
  - References:
    - HasDescription: `Application.CommandRequest`
    - HasTypeDefinition: `DataTypeEncodingType`
- Description node (|desc|)
  - NodeClass: `Variable`
  - DataType: `String`
  - Value: `Application.CommandRequest`

Debug print of DescriptionNode ReferenceDescription:
```
DescNode: [ReferenceDescription { reference_type_id: NodeId { namespace: 0, identifier: Numeric(39) }, is_forward: true, node_id: ExpandedNodeId { node_id: NodeId { namespace: 4, identifier: String(UAString { value: Some("|desc|CODESYS Control for Linux SL.Application.CommandRequest") }) }, namespace_uri: UAString { value: None }, server_index: 0 }, browse_name: QualifiedName { namespace_index: 4, name: UAString { value: Some("Application.CommandRequest") } }, display_name: LocalizedText { locale: UAString { value: None }, text: UAString { value: Some("Application.CommandRequest") } }, node_class: Variable, type_definition: ExpandedNodeId { node_id: NodeId { namespace: 0, identifier: Numeric(69) }, namespace_uri: UAString { value: None }, server_index: 0 } }]
```

Screenshots of UAExpert for inspected nodes:
[image]

[image]

[image]

[image]

Do you have some idea on how UAExpert may be able to load these structs and read/write to them from this information? 
Is there anywhere else I can inspect/browse to provide more information?

einarmo, 2025-08-06:
Ah, I think I get it. This: https://reference.opcfoundation.org/Core/Part5/v104/docs/D, I think. We can look into supporting this, though it _is_ a deprecated way to do this.

porquey, 2025-08-06:
I think that would be nice. But doesn't sound like small piece of work.

Updating `StaticTypeLoader` and `TypeLoaderInstance` to take generic NodeId might be a more favorable approach. I was able to get something working by hacking it to take non Numeric nodes, but I assume the codegen module also needs to be updated.

porquey, 2025-12-15:
Hello,

I have confirmed that it is indeed the case, that Codesys server generates OPC UA server that relies on the DataTypeDictionary method of defining custom enums and structs, as described here: https://documentation.unified-automation.com/uasdkcpp/2.0.1/html/L2DataTypes.html#L2DataTypesDictionary

Inspecting the OPCUA server node tree:
```
Root
├ Objects
├ ...
└ Types
  ├ ...
  └ DataTypes
    ├ BaseDataType
    ├ XML Schema
    └ OPC Binary
      ├ CODESYSSPV3/3S/IecVarAccess
      ├ Opc.Ua.Di
      └ Opc.Ua
```

Reading the DataType Dictionaries above I see these type of XML data:
```
<opc:TypeDictionary 
    xmlns:xsi='http://www.w3.org/2001/XMLSchema-instance' 
    xmlns:tns='CODESYSSPV3/3S/IecVarAccess' 
    DefaultByteOrder='LittleEndian' 
    xmlns:opc='http://opcfoundation.org/BinarySchema/' 
    xmlns:ua='http://opcfoundation.org/UA/' 
    TargetNamespace='CODESYSSPV3/3S/IecVarAccess'>
    
    <opc:Import Namespace='http://opcfoundation.org/UA/'/>
    
    <opc:StructuredType BaseType='ua:ExtensionObject' Name='Application.CommandInterface'>
        <opc:Field TypeName='tns:Application.CommandRequest' Name='command'/>
        <opc:Field TypeName='tns:Application.CommandResponse' Name='response'/>
    </opc:StructuredType>
    
    <opc:StructuredType BaseType='ua:ExtensionObject' Name='Application.CommandResponse'>
        <opc:Field TypeName='opc:CharArray' Name='command'/>
        <opc:Field TypeName='opc:CharArray' Name='command_id'/>
        <opc:Field TypeName='opc:CharArray' Name='command_state'/>
        <opc:Field TypeName='opc:CharArray' Name='message'/>
        <opc:Field TypeName='opc:UInt32' Name='reason'/>
    </opc:StructuredType>
    
    <opc:StructuredType BaseType='ua:ExtensionObject' Name='Application.CommandRequest'>
        <opc:Field TypeName='opc:CharArray' Name='command'/>
        <opc:Field TypeName='opc:CharArray' Name='command_id'/>
        <opc:Field TypeName='opc:CharArray' Name='command_source'/>
    </opc:StructuredType>
    
</opc:TypeDictionary>
```

oroulet, 2025-12-17:
less and less server use that. That format i cumbersome and not so easy to implement correctly. I would just wait for someone who really want it to make a MR